MosaicManufacturing/canvas-issues #264: Import printer-profile created in Simplify3d v5.01 failed

Muratore73, 2023-02-17:
**Describe the bug**
The import of printer-profiles created with new Simplify3D v5.01 fail with error: "Cannot read properties of null (reading 'replace')"

**To Reproduce**
Steps to reproduce the behavior:
1. Log on to https://www.canvas3d.io/
2. Click on 'printers'
3. Click on '+ New Printer'
4. Select 'Import'
5. Select file to import

**Expected behavior**
Import printer-profile an create new printer in Canvas.

**Operating System (please complete the following information):**
 - OS: Windows 11
 - Browser: Edge 
 - Version: Version 110.0.1587.46 (Offizielles Build) (64-Bit)

**Additional context**

brandonbloch, 2023-02-17:
Hi @Muratore73,

Thanks for reporting this issue! We were made aware of this problem a few days ago and realized that while the underlying format of S3D profiles hasn't changed in v5 (i.e. XML files with .fff extension), the actual content is significantly different. We're working on a fix for this in order to support importing these profiles, but unfortunately there isn't a good workaround in the meantime other than manually creating a Canvas profile with the same settings.

I'll leave the issue open until full support for S3D v5 import is available in Canvas.

Cheers,
Brandon

Muratore73, 2023-02-23:
Hi Brandon

Thanks for your support on this.
If can help testing, or provide some S3D-profiles, let me know.

Kind regards
Christoph

brandonbloch, 2023-03-01:
Hi @Muratore73, thanks for the offer! We do have a couple of profiles we're using to test the new logic, but if you have any profiles you're willing to post here that would certainly be appreciated, particularly ones with different/non-standard variations of configuration. If you're not willing to post them publicly but are fine with us adding them to our collection of test profiles, you can send them my way at brandon@mosaicmfg.com.

Cheers,
Brandon

a55830, 2023-03-15:
[CrealityCR-6_max.zip](https://github.com/MosaicManufacturing/canvas-issues/files/10981525/CrealityCR-6_max.zip)
hope you can use this, i realy need that to work

brandonbloch, 2023-03-17:
Thanks @a55830, I'll include this in our test suite. We expect to have this feature available next week.

Cheers,
brandon

closed 2023-03-21 by brandonbloch

Muratore73, 2023-03-21:
> Hi all, support for S3D v5 profile import is now available! I'll close this issue as completed; if you have any other questions feel free to continue commenting in this thread, or if you're still experiencing errors feel free to re-open it.

Hi Brandon

Import for S3Dv5-Profile works perfectly now.
Thanks for this great support!!!

Kind regards
Christoph