MosaicManufacturing/canvas-issues #267: Feature Request: I would like to see canvas support .3mf files

Mightywiz, 2023-08-19:
A lot of new Color files are coming out for the Bambo labs printers, it would be nice to make those .3mf files compatible with the canvas software for printing.

If they are already colored and available for download why should we have to recreate the file and waste the time to color it ourselves.

this would make canvas a greater program for all to use.

thank you

Strell69, 2023-10-18:
There are Tribe files that use .3mf with colors already added saving lots of time having to color manually in Canvas. Adding .3mf is very useful.

chronoite, 2023-10-22:
Converting between the stl and 3mf can be problematic as the converted model may have faceted curves, if you don’t do any manual clean-up. (Since 3mf, like stl, is a polygon-based file format); so it would be ideal to have support for more file types

wkuricker, 2023-11-08:
Agreed.  This feature would be incredibly useful.

moberli, 2023-11-25:
Agree - 3mf is coming up more and more..

vegetablejedi, 2024-07-07:
Agree. Canvas needs to provide more features, and support for various file formats is one of them.