MosaicManufacturing/canvas-issues #268: Link to "Open-source licenses" points to a 404

robzach, 2023-09-07:
**Describe the bug**
The link to read the license text for all the open-source projects included in Canvas points to a 404.

**To Reproduce**
Click "Open-source licenses" in the lower left of Canvas (I'm accessing it at CMU so the URL mine points to is `https://array.ideate.cmu.edu/licenses.txt`)

**Operating System (please complete the following information):**
 - OS: OSX 13.5
 - Browser: Chrome
 - Version: 116.0.5845.140

brandonbloch, 2023-09-07:
Hi @robzach, thanks for reporting! The link should actually be https://canvas3d.io/licenses.txt – we should have this fixed shortly.

Thanks,
Brandon

brandonbloch, 2023-09-08:
Hi @robzach, this should now be fixed. Thanks again for letting us know.

Happy printing!
Brandon