MosaicManufacturing/canvas-issues #274: Slic3r leading zero missing in Z movement

moberli, 2023-11-17:
**Describe the bug**
If you use Slic3r instead of KISSlicer - the files contain a Z movement without leading Zero example:
G1 Z.8 F3000 ; lift Z

It will be glad if there is a possibility to activate the Regex function to allow leading zero's 

**To Reproduce**
Steps to reproduce the behavior:
1. Slice a Object with Slic3r and see the G1 Z movement.

**Expected behavior**
It will be glad if in the option is possible to activate leading Zero's with the solution Prusa had added to theirs Slic3r.

**Additional context**
https://github.com/prusa3d/PrusaSlicer/issues/6996

brandonbloch, 2023-11-22:
Hi @moberli,

Thanks for linking to that issue! We've added a postprocessing step that should restore leading zeros as needed, so this should now be addressed and working as you'd expect. If you're still running into any issues, please feel free to comment on this thread or re-open the issue.

Cheers,
Brandon

closed 2023-11-22 by brandonbloch

moberli, 2023-11-25:
Great Job - Thank you!