MosaicManufacturing/canvas-issues #278: Error 500 from Palette3pro at sending gcode

HarlekinSimplex, 2024-06-11:
**Describe the bug**
Today I placed two different .STL files into canvas to print them with just one color with my Pallete3pro connected to a Snapmaker Artisan.
Coloring and slicing the two models with the same color went fine.
But as soon as I send the result to the device Canvas received an Error 500 from the device.
As soon as I used two colors (one for the first and a different color for the 2nd) sendig the data to the Pallete3Pro was successfully completed as usual.   

**To Reproduce**
Steps to reproduce the behavior:
1. Load two .STL files into Canvas
2. Use a Palette3pro as Device 
3. Give both objects the same color (I used #1)
4. Splice the print
5. Click 'Send to Device'
6. See the red Error Box in the right corner of the window showing 'Error 500 received from Device'

**Expected behavior**
A clear and concise description of what you expected to happen.
I would like to use Canvas and Pallete3Pro for all my single extruder prints, regardless if I run a single or multicolor print.

**Screenshots**
The p3p should accept and print monochrome prints like it accepts multicolor prints.
Just without splicing filament.

**Operating System (please complete the following information):**
 - OS: MacOs and Window10
 - Browser: Brave and Firefox
 - Most Recent (May 24)

**Additional context**
Add any other context about the problem here.

brandonbloch, 2024-06-11:
Hi @HarlekinSimplex,

For the same reasons mentioned in #279, this isn't supported by Palette 3 or Palette 3 Pro at this time, which is why this error is occurring. Canvas should not be allowing users to send single-color/single-material projects to their Palette, which is a bug that we'll look to fix, but this functionality can't be added to Canvas without support on the Palette side.

Cheers,
Brandon

brandonbloch, 2024-07-08:
Hi @HarlekinSimplex, the bug that inadvertently allows sending single-color files to Palette 3 (which isn't supported) has been fixed in this morning's Canvas update.

Thanks,
Brandon

closed 2024-07-08 by brandonbloch