MosaicManufacturing/canvas-issues #280: Objects listed in the object window don't appear on the slicing table.

rmhoward33322, 2024-07-22:
It appears Auto Arrange has changed. When I attempt to duplicate objects and the Auto Arrange determines there's no space on the slicing table for the number of objects I want, the item gets added to the object window but doesn't appear anywhere on or off the slicing table. I have plenty of room on the slicing table for another object but when I ask to make a duplicate, i get a message that there were 2 solutions. In the bottom right it indicates there may not be enough space. OK, but where is the object? the object is added to the Object list, but is nowhere to be seen. And if there are 2 solutions, how do i see the 2 solutions. I went so far as to duplicate my object with 2 new objects and they worked fine. I arranged the 3 objects on the screen. I then moved one of the objects to a new location on the slicing table, leaving the space open for another object, but when I try to duplicate the object again, it adds the object to the object list (item #4), but the object doesn't appear anywhere that I can see. I read the release for July 8 on new auto arrange features, where can I go to get help on how they work? Perhaps this isn't a bug, I just don't know how to use the new features. If so, I need some kind of help on how to use it. 

**To Reproduce**
Steps to reproduce the behavior:
1. I started a new project and added my print to the slicing table.
2. I resized the print to 115% of the original size
3. I painted the object
4. I asked to duplicate the object with 5 new objects.
5. The lower right corner of the screen indicated the objects may not fit.
6. only one object appears on the slicing table although the object list indicated there are 6 objects. Where are the other 5?
7. A message appeared while Auto Arrange was trying to arrange the items and indicated to press Stop if you want to manually arrange them, but that did no good. I pressed stop and it stopped, but still only shows one object on the slicing table but lists 6 items in the object list.

**Expected behavior**
If the objects don't fit the slicing table, the "extra" objects should appear on the screen off the table so I can determine what to do. Manually arrange them myself or delete them if they truly don't fit on the table..

**Screenshots**
[image]
in this example I was able to create 3 objects, moved one of the objects to a new place on the slicing table leaving room for another object. when I asked to duplicate object #1 object 4 was added to the list, but doesn't show up; anywhere.

**Operating System (please complete the following information):**
 - OS: Windows 11 Pro
 - Browser Edge
 - Version          of what? Windows? 23H2 Browser? Version 126.0.2592.102 (Official build) (64-bit)

**Additional context**
I am in need of being able to duplicate the objects as I need them. This is causing me much grief.,, as I have a deadline to get these prints finished. and it's fast approaching.  I wish it was easier to learn about changes you make, especially when the changes just show up without warning.

brandonbloch, 2024-07-23:
Hi @rmhoward33322,

Sorry to hear you're running into issues with the new auto-arrange feature. There are a few things I want to clarify to make sure we understand the issue fully:

- When you duplicate a model (or a group of models), the initial position of the new models before they are arranged is in the same spot as the originals. When you say the new objects don't appear, are they overlapped with the existing models or not present at all? You can tell if they're actually just overlapped by using the Move tool – moving just one of the new models away should show the original model in that same position
   - If the new models aren't present on the print bed after trying the above, there is likely a front-end bug involved here, but refreshing the page to re-load the models should work around this issue in the short term
- The "Stop" and "Cancel" buttons behave differently from one another – "Stop" will use the best solution found so far (which will always be a valid auto-arrange result) and "Cancel" will abort the operation entirely, leaving the duplicated models in their starting positions (where they would be overlapped with the originals). Are you sure you used "Stop" and not "Cancel"? If so, this may be a bug in the new feature, but if you used "Cancel" this should actually be the expected result
- While the new auto-arrange algorithm is running, you can simply let it proceed and apply the best solution without stopping or cancelling. Have you tried that, and if so, did you run into any issues?

Let us know when you get a chance; these clarifications will help us figure out what's going on and fix any bug that might be present in the new implementation.

As for making it easier to learn about updates in Canvas, this is something we're working on, and we're aiming to make new features and improvements more discoverable directly within the application.

Cheers,
Brandon

brandonbloch, 2024-07-25:
Hello again @rmhoward33322,

Thank you for your patience. We've investigated the issue more deeply and indeed found an edge case bug in the new auto-arrange algorithm related to circular print beds. This bug could result in incorrect handling of "overflowed models" when not all models could be placed on the bed automatically, as well as potentially incorrect model placement attempts in specific cases even when all models should be able to be placed.

This bug fix has just been released, and our testing indicates that the issue you're experiencing should now be resolved, and auto-arrange should be working as expected. If you're still experiencing any issues, please don't hesitate to comment on or re-open this issue.

We hope that with the issue resolved, the new auto-arrange implementation makes your life easier (rather than harder!) and overall produces better arrangements than the old algorithm, especially with circular print beds.

Good luck with your deadline and happy printing!

Cheers,
Brandon

closed 2024-07-25 by brandonbloch