MosaicManufacturing/canvas-issues #281: unable to change printing material

supplyandsupport, 2024-09-19:
I have downloaded STL projects and when I process them in Canvas , I cant seem to change the print media from lets say, ABS to PLA, if I drag the PLA to the model, then when I go to print, it has both ABS and PLA being printed at the same time, its like the ABS is hard coded into the STL file, how can I clear the ABS and choose my own printing material?

brandonbloch, 2024-09-19:
Hi @supplyandsupport, sorry to hear you're running into issues when slicing. Based on your description, a few questions come to mind:

- Is your project configured for printing with Palette, Element, or neither?
- When you say the print has both ABS and PLA being printed at the same time, what exactly do you mean by this?
   - Are some of the paths in the print preview using ABS and others are using PLA?
   - Are there duplicate or overlapping paths?
- Do you have supports or rafts enabled on your project? The input that they use (assuming you're using Palette or Element) can be configured via the project settings UI

Would you be willing to share your project either here or to support@mosaicmfg.com? This would help us figure out the issue you're running into more quickly. Otherwise, would you mind including a screenshot of the print preview and summary view?

Cheers,
Brandon

supplyandsupport, 2024-09-20:
I have downloaded a file from thingiverse and was trying to print in PLA, it seemed to be setup to be printed in ABS. I am using a Mosaic Element printer. I want to print the item in PLA not ABS, so I try by dragging the PLA onto the model, but then it will only add the PLA, not replace ABS, so how can I change the material from ABS to PLA for this model?
[image]
[image]

brandonbloch, 2024-09-20:
Thanks for the additional info – based on this, I'd expect that this is related to support settings for the project. If you open project settings in the top-right of the UI, do you have supports and/or interfaces enabled and configured to use Input 1? This is the default for Element profiles:

[image]

If that's the case, there are a few possible ways to resolve this:
- Disable supports and/or support interfaces if they're not needed
- Switch supports to Auto and support interfaces to use Input 7 (the input you have assigned to PLA)
- Use Input 1 for PLA and leave support settings as-is

Let me know if that helps!

supplyandsupport, 2024-09-20:
That did not work, I this project does not have supports, and it seems that ABS is hardcoded into the file, so thank you for trying

Screenshot.2024-09-20.at.11.42.05.AM.png (view on web)<https://github.com/user-attachments/assets/4abaac87-3b80-49a9-991b-e9346fe35195>

brandonbloch, 2024-09-20:
@supplyandsupport models don't contain direct references to material data; this should be based only on any assigned materials (including paint data, if the model is painted). Would you be able to provide a share link for this project so we can investigate further? You can either post it here, or if you prefer to keep it private, you can send it to support@mosaicmfg.com

brandonbloch, 2024-09-20:
Do you have a raft enabled on the project?