MosaicManufacturing/canvas-issues #283: Every project I try to slice gets internal server error 3008

Core3dTech, 2025-02-14:
Every project that I try to slice causes the same error. all projects have been successfully sliced and printed in previous weeks

brandonbloch, 2025-02-18:
Hi @Core3dTech, really sorry to hear you're running into this issue. We released some under-the-hood improvements to the slicer last week, and we've tested the changes fairly extensively but it seems like there's something we missed that's causing this issue. Would you be able to share an example project and either post the sharing link here or send it to brandon@mosaicmfg.com? If we're able to dig into the issue we should be able to investigate and hopefully resolve it fairly quickly. Please make sure to include all profiles etc. by checking all the boxes:

[image]

Thanks in advance, and apologies again for the inconvenience.

Cheers,
Brandon

Core3dTech, 2025-02-18:
I nailed the problem down to the z-travel speed. It might be a new field and it defaults to x/y speed. If I lower it to 10 everything slices again. Thank you,Maarten Sent from my iPhone

Hi @Core3dTech, really sorry to hear you're running into this issue. We released some under-the-hood improvements to the slicer last week, and we've tested the changes fairly extensively but it seems like there's something we missed that's causing this issue. Would you be able to share an example project and either post the sharing link here or send it to ***@***.***? If we're able to dig into the issue we should be able to investigate and hopefully resolve it fairly quickly. Please make sure to include all profiles etc. by checking all the boxes:
Screenshot.2025-02-18.at.12.41.25.PM.png (view on web)

brandonbloch, 2025-02-18:
Glad to hear things are working again! We're not entirely sure why this change would resolve the 3008 error (we'd expect the projects to still slice, albeit with higher feedrates for Z movement) so if you'd be willing to share a project that's still having this issue, we'd definitely appreciate it! Regardless, I'll close this issue since you've been able to resolve it.

Thanks,
Brandon

closed 2025-02-18 by brandonbloch